espressif/ESP8266_NONOS_SDK #271: 最新版本xtensa-lx106-elf编译器无法编译NONOS-SDK

liruya, 2019-10-22:
最新版本xtensa-lx106-elf-linux64-1.22.0-100-xxxxxx-5.2.0编译器无法编译NONOS-SDK，
`In file included from ../../include/os_type.h:28:0,
                 from ../../include/user_interface.h:28,
                 from app_common.h:4,
                 from user_uart.h:5,
                 from user_uart.c:1:
user_uart.c: In function 'uart_enable_isr':
../../include/ets_sys.h:96:5: error: implicit declaration of function 'ets_isr_unmask' [-Werror=implicit-function-declaration]
     ets_isr_unmask((1<<inum))
     ^
../../include/ets_sys.h:102:5: note: in expansion of macro 'ETS_INTR_ENABLE'
     ETS_INTR_ENABLE(ETS_UART_INUM)
     ^
user_uart.c:71:2: note: in expansion of macro 'ETS_UART_INTR_ENABLE'
  ETS_UART_INTR_ENABLE();
  ^
user_uart.c: In function 'uart_disable_isr':
../../include/ets_sys.h:99:5: error: implicit declaration of function 'ets_isr_mask' [-Werror=implicit-function-declaration]
     ets_isr_mask((1<<inum))
     ^
../../include/ets_sys.h:105:5: note: in expansion of macro 'ETS_INTR_DISABLE'
     ETS_INTR_DISABLE(ETS_UART_INUM)
     ^
user_uart.c:75:2: note: in expansion of macro 'ETS_UART_INTR_DISABLE'
  ETS_UART_INTR_DISABLE();
  ^
cc1: all warnings being treated as errors
make[1]: *** [../../Makefile:352：.output/eagle/debug/obj/user_uart.o] 错误 1`

使用xtensa-lx106-elf-linux64-1.22.0-88-xxxxxx-4.8.5可以编译，如何解决？

xcguang, 2019-10-25:
你用的是最新版的 SDK 吗？我们应该修复过一次

liruya, 2019-11-05:
> 你用的是最新版的 SDK 吗？我们应该修复过一次

用的NONOS-SDK 3.0

Town-Young, 2020-02-08:
我也遇到了这个情况，最后选择用安信可IDE v0.5版本，可以正常编译