espressif/ESP8266_NONOS_SDK #272: Request to increase buffer in AT+CIPSSLSIZE

aminismail, 2019-10-23:
Please consider adding the capability to specify a SSLSIZE of 8192 using the AT+CIPSSLSIZE command. Several users have confirmed that it is stable when they modified the code to increase the buffer size. Without this it is impossible to use the ESP8266 as an MQTT client using TLS. One of the important applications of the ESP8266 is in IoT and now most MQTT servers allow only TLS connections. The ESP8266 is a great product and with this capability it will be even better. 
Thank you.

xcguang, 2019-10-23:
👌
Tomorrow，we will update it.

aminismail, 2019-10-23:
Super... Thank you.

aminismail, 2019-10-24:
Hi,
  I downloaded and flashed the SDK and I am still getting an error:

AT+GMR
AT version:1.7.0.0(Aug 16 2018 00:57:04)
SDK version:3.0.0(d49923c)
compile time:Aug 23 2018 16:58:12
Bin version(Wroom 02):v1.7.0
OK
AT+CIPSSLSIZE?
+CIPSSLSIZE:2048
OK
AT+CIPSSLSIZE=8192
ERROR

xcguang, 2019-10-25:
Do you test it using official firmware? The newest feature isn't released, you can compile it by yourself, the range is (2048,8192), 

the code is 
```
#define ESPCONN_SECURE_MAX_SIZE 8192
bool ICACHE_FLASH_ATTR espconn_secure_set_size(uint8 level, uint16 size)
{
	if (level >= ESPCONN_MAX || level <= ESPCONN_IDLE)
		return false;

	if (size > ESPCONN_SECURE_MAX_SIZE || size < ESPCONN_SECURE_DEFAULT_SIZE)
		return false;

	max_content_len = size;
	return true;
}
```

aminismail, 2019-10-25:
OK... I see I did not realize that. Thanks!

aminismail, 2019-10-25:
Hello xcguang,
  Are you planning a new release with this change in the near future? If so, I would rather wait then try to do the build myself since I am not too familiar with the tools or the process.
Thanks!

xcguang, 2019-10-25:
Now we are busy with esp32 and esp8266 AT release. So, NONOS SDK may not be released in the near month. Sorry for the inconvenience.

aminismail, 2019-10-25:
I definitely understand xcguang. Is it possible for you to do a compile/build for us so we can update just the changed binaries and test it out.
You see, we have several wireless sensor nodes out in the field that use the ESP8266 and we are now required to use SSL/TLS since we have to switch over to AWS which only supports TLS over MQTT. So it would certainly help us if we could try out the change you made to see if it works properly.
Thank you for your consideration.

xcguang, 2019-10-28:
[user1.2048.new.5.bin.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/3776769/user1.2048.new.5.bin.zip)

I didn't test it, if you have any question , please just feel fret let me know, thanks.

aminismail, 2019-10-28:
Thank you for doing this for us xcguang. We updated the firmware and attempted to try it out. However it did not work since the CIPSSLSIZE=8192  is not saved. So on reset or power up, the buffer is set to the default.

aminismail, 2019-10-28:
** UPDATE **
Since UART-WiFi Passthrough mode is not supported with SSL, we tested the ESP8266 in Normal mode (CIPMODE=0) and everything works fine. Thank you.

I would like to suggest that in a future update of the SDK that you support UART-WiFi Passthrough Mode with SSL client and not just TCP client. All our devices use this mode and I am sure a lot of other users also use it. The ESP8266 works great in this mode with TCP connections and if SSL could be supported that would be excellent.

 Thank you once again for your excellent support and help. Hope to see my suggestion implemented in a future release.

xcguang, 2019-10-29:
Hi,
1. You can try `AT+CIPSSLSIZE=8191`.
2. We have no plan to add UART-WiFi Passthrough Mode with SSL client in the nonos

If possible, you can try our new ESP8266 AT, which is compatible with ESP32 AT. And NONOS SDK will be dead soon, we will focus on ESP8266 RTOS AT.

aminismail, 2019-10-29:
OK. I will investigate the ESP8266 RTOS AT. If it has a feature similar to UART-WiFi Passthrough mode, we can definitely switch over to using it instead of the NONOS.
Thanks!

xcguang, 2019-11-06:
@aminismail 
We have a comparison table in Chinese. 
https://blog.csdn.net/espressif/article/details/101208054

aminismail, 2019-11-06:
Yes... I think it will work for us. I would like to go ahead and flash the AT firmware onto our devices. Are the binaries available? I could not find them. Could you please provide me with a link to the binaries and instructions to flash the ESP8266
Thanks.

xcguang, 2019-11-06:
You can download firmware `ESP8266 IDF AT Bin V2.0` at https://www.espressif.com/zh-hans/support/download/at, please read https://github.com/espressif/esp-at/blob/master/docs/ESP_AT_Get_Started.md at first, the AT port pin is different from AT v1.7