espressif/ESP8266_NONOS_SDK #275: ESP no longer connects in SSL mode since last mbedtls updates

davydnorris, 2019-11-05:
I just pulled the latest updates and recompiled my application, and none of my SSL connect calls work any more, failing with certificate handshake errors.

I double rebuilt the CA certs and still no luck. Rolling back to the older SDK (prior to commit #08957d1 on 30 Sep) works fine.

Have there been any changes to what certificates are accepted or the certificate code? I am getting several errors but many seem to be related to the PK module:

HTTPS connect to google results in:
```
client handshake start.
espconn_mbedtls.c 662, type[TLS.ca_x509.cer],length[1354]
E:M 536
E:M 536
client handshake failed!
Reason:[-0x2700]
INFO : HTTPCLIENT: disconnected with error, code = -39
```
HTTPS connect to ibmcloud results in:
```
client handshake start.
espconn_mbedtls.c 662, type[TLS.ca_x509.cer],length[1338]
E:M 536
client handshake failed!
Reason:[-0x4290]
INFO : HTTPCLIENT: disconnected with error, code = 46
```

DNS and SNTP connections are all working fine, it's just the SSL ones that are not.

davydnorris, 2019-11-07:
I have done some more playing and it's definitely the mbedtls library - pulling the current dist and replacing mbedtls with my current working copy gives me a working build of my system.

I have also rolled back to the very next commit after my working lib and tried using that, and it fails. So the commit that breaks my code is #08957d1 on 30th Sept.

Is it possible to get more details on what changed in this build?

kriegste, 2019-11-07:
Some options were switched on
https://github.com/espressif/ESP8266_NONOS_SDK/commit/08957d11d2456a8521fa471fb216cf9ec8dedb5a
You can try them one by one and recompile mbedtls, but I think it is just an out-of-memory error...

kriegste, 2019-11-07:
I wonder why ssl v3 was activated in the first place. Soon it is 2020, and ssl v3 is deprecated anyway...
https://tools.ietf.org/html/rfc7568

xcguang, 2019-11-07:
Hi,
Because many companies still use SSL V3, we continue to receive feedback that they are not able to connect to their own servers normally.

davydnorris, 2019-11-07:
> 
> 
> Some options were switched on
> [08957d1](https://github.com/espressif/ESP8266_NONOS_SDK/commit/08957d11d2456a8521fa471fb216cf9ec8dedb5a)
> You can try them one by one and recompile mbedtls, but I think it is just an out-of-memory error...

I've cranked the SSL buffer up to max and it still not playing nice - previously was working fine with 5120 bytes so if an extra 3k doesn't do it there's something wrong.

Oh and now I've noticed another thing - it's not coming back from deep sleep or a reset any more, and it's difficult to reflash when that happens. Think there's stuff in the other libraries that isn't good either.

Looks like I'll just roll back to the stable version prior to Sep 30

davydnorris, 2019-11-07:
> 
> 
> Hi,
> Because many companies still use SSL V3, we continue to receive feedback that they are not able to connect to their own servers normally.

Maybe because it's deprecated and a security problem, it would be better to leave it off by default but give them the option to recompile with it on?

BTW I have now reverted to a baseline of all libraries just before you turned SSL3.0 on and have recompiled, and my code is working fine - connecting properly (with SSL memory back to 5120) and coming back from sleep and reset properly.

Coming back from a reset I would get a cause 2 and then almost straight away a cause 4, and then after a little bit I would get a wtd and nothing after that.

kriegste, 2019-11-07:
Are those other problems mbedtls-related, too?

davydnorris, 2019-11-07:
nope - deep sleep and reset are done after the wifi is shut down. I pulled the latest version of the libs and then replaced mbedtls with the version that worked. After rebuilding the unit didn't come out of deep sleep properly - I can actually see it start with info at 78k baud but then it dies. I'm running at 115k so it's garbage chars - I may have to drop everything down to 78k to see what it's doing.

When I go back to my baseline it's all solid again.

I had a similar issue previously which I solved by adding an os_delay_us(100000) just after the deep sleep - seemed it needed some quiet time to think before going to sleep. I've increased that, put delays in before and after, tried a bunch of things and no luck - the new libs don't like deep sleep in my code.

kriegste, 2019-11-07:
Have you tried swapping only libmain? To find out which commit exactly causes this?
Another hint would be to retrieve the exact boot message after the crash (adjust the baud rate prior to it). The given address can then be looked up in the map file and might help Espressif.

davydnorris, 2019-11-07:
Yeah I've worked my way through the commits but still nothing concrete apart from mbedtls not working from the first commit after my current stable SDK.

Dropping back to 78k baud is my next move to try and see what's going on. It's very weird.

fvpalha, 2019-11-08:
@davydnorris What is your stable SDK?

davydnorris, 2019-11-09:
OK @fvpalha this is for me - you may have a better experience, but I have been using the SDK current as of the commit before 30 September.

I have found that using current as of now and replacing libmbedtls with the one before 30 September is OK, but I am noticing that I'm not coming back from deep sleep properly. If you don't use deep sleep then you may be OK. I am also looking into other factors in my code and power supply to rule them out - I have seen these symptoms before but they are random, and in the current SDK they seem much more consistent so it may be that the latest libs are highlighting a problem in my unit or code

davydnorris, 2019-11-09:
@kriegste so I changed my baud rate within my app to 74880 and just went through a cycle, and everything looks normal except it doesn't come back properly from deep sleep - it did wait the correct amount of time, and it did try to boot, but stopped there. What's interesting is that I would expect rst cause to be 5, not 2!

@xcguang - can you have a look too please? Any help very welcome.

PS: I use system_deep_sleep(), not system_deep_sleep_instant() 

```
INFO : TIMER: platform is idle, try to sleep
INFO : SYSINFO: write volatile store,  len = 152
INFO : RUNTIME: shutting down
INFO : IOTF: disconnecting...
INFO : SENSOR: sensor system shutting down
INFO : RUNTIME: Runtime stopped: false, Wifi stopped: 4
INFO : TASKS: Received event on queue 0: 0x05000000 0x3fff35c8
INFO : MQTT_TCP: Disconnected callback
INFO : IOTF: disconnected
INFO : RUNTIME: disconnected
INFO : WIFI: closing connection...
state: 5 -> 0 (0)
rm 0
del if0
usl
mode : null
INFO : WIFI: event: STAMODE_DISCONNECTED
INFO : WIFI: disconnect ssid NorrisNet, reason 8, ASSOC_LEAVE
INFO : WIFI: event: OPMODE_CHANGED
INFO : RUNTIME: sleeping for 294869 milliseconds
▒▒ter deep sleep
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

davydnorris, 2019-11-09:
The one thing I have noticed in my searches on this is that a lot of times the cause is due to not having pullups on GPIO0 and GPIO2. I was pretty sure the Lolin modules had pullups but I double checked and they have 12k on both.

Do you reckon this might be too weak a pullup? I could try adding another 10k or 12k resistor externally to make it stronger I guess. Weird that it has only recently stopped coming out of sleep though

awilhite, 2020-03-12:
I'm running out of IRAM if I use the libmbedtls.a from before the commit on 30 September with SDK 3.0.2. Does anyone have instructions on how to rebuild SDK libraries using esp-open-sdk, or can someone provide a libmbedtls.a built with SDK 3.0.2 that has the offending commit removed?

kriegste, 2020-03-12:
If you only need the client and not the SSL server part, you can try this:
https://github.com/kriegste/ESP8266_NONOS_SDK/raw/master/lib/libmbedtls.a

awilhite, 2020-03-13:
@kriegste I tried the compiled mbedtls library in your repository, but I found that all of the mbed functions are placed into the .text section (IRAM), whereas the library in espressif repository has all of the mbed functions placed in irom0.text (flash). Because of this, I get the "`.text' will not fit in region `iram1_0_seg'" message during linking. Any ideas why this might be?

kriegste, 2020-03-13:
Do you use your own linker file? Or the original one?

The original does this:
```
  .irom0.text : ALIGN(4)
  {
[...]
    *libmbedtls.a:(.literal.* .text.*)
```

No problems here.

awilhite, 2020-03-14:
@kriegste I figured out what I did wrong. I had renamed the library you had given me to differentiate it from the original, not realizing that the linker depended on the name being correct in order to place it in flash. I reverted the name change and all is well. Thanks for your help!

kriegste, 2020-03-14:
It should save even a bit more flash and RAM compared to official versions before September 30th.