espressif/ESP8266_NONOS_SDK #278: 请求工程师帮忙解答这边三个疑问

shiloye, 2019-12-04:
### 1：add if0 后面这里是在做什么动作呢？为什么station_connect需要扫描信道3s左右

----------wifi_station_connect------
free heap =62376
mode : sta(5c:cf:7f:26:84:fb)
add if0

[09:44:21.808]收←◆scandone
state: 0 -> 2 (b0)

[09:44:21.836]收←◆state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 8
cnt _

### 模块直接供电时候，add if0会卡3s，但是在已经连上网情况下，进行外部复位，add if0 后面只用了0.3S 就scandone了

[09:44:21.610]
##wifi channel:1
##static ip:192.168.1.5
------------------save_flag:0xBB---------------------
did=00000064
----------wifi_station_connect------
free heap =62376
mode : sta(5c:cf:7f:26:84:fb)
add if0

[09:44:21.808]收←◆scandone
state: 0 -> 2 (b0)

[09:44:21.836]收←◆state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 8
cnt 

### 3. 我用API指令去指定信道扫描，100ms即可扫描到我要的路由信息，此时我如何指定模块连接他但不要去做全信道扫描？我下面的代码为什么没有生效
`
static void ICACHE_FLASH_ATTR scan_done(void *arg, STATUS status)
{
	DBG("enter the scan_done \n");
	wifi_set_opmode(STATION_MODE);
	wifi_set_phy_mode(1);
	uint8 ssid[33];
	char buffer[256];
	if (status == OK)
	{
		struct bss_info *bss_link = (struct bss_info *)arg;
		struct station_config inf;
	   
		os_printf("status:%d pointer:%p\n", status, arg);
		while (bss_link != NULL)
		{
			os_memset(ssid, 0, sizeof(ssid));
			os_memset(buffer, 0, sizeof(buffer));
			if (os_strlen(bss_link->ssid) <= 32)
			{
				os_memcpy(ssid, bss_link->ssid, os_strlen(bss_link->ssid));
			}
			else
			{
				os_memcpy(ssid, bss_link->ssid, 32);
			}
			os_sprintf(buffer, "+PSCAN:(%d,\"%s\",%d,\"" MACSTR "\",%d,%d,%d)\r\n", bss_link->authmode, ssid, bss_link->rssi, MAC2STR(bss_link->bssid), bss_link->channel, bss_link->freq_offset, bss_link->freqcal_val);
			os_printf("scan result:%s\n", buffer);
		
			os_memcpy(&inf.ssid,ssid,os_strlen(ssid));	
			os_memcpy(&inf.password,DEFAULT_PWD,os_strlen(DEFAULT_PWD));	

            wifi_fast_scan_threshold_t fast_scan;
	        // fast_scan.rssi=bss_link->rssi;
			fast_scan.authmode=bss_link->authmode;
			inf.threshold=fast_scan;
			inf.bssid_set = 0;
			wifi_station_set_config_current(&inf);
			wifi_station_connect();
			// wifi_station_set_auto_connect(TRUE);
			bss_link = bss_link->next.stqe_next;
		}
		os_memset(buffer, 0, sizeof(buffer));
		os_sprintf(buffer, "success status=%d\r\n", status);
	}
}
 `

ustccw, 2020-01-11:
1. `add if0` 后面在全信道扫描
2. 内部已经保存过信道，不需要全信道扫描
3. 连接前需要设置信道有效值

shiloye, 2020-01-16:
> 1. `add if0` 后面在全信道扫描
> 2. 内部已经保存过信道，不需要全信道扫描
> 3. 连接前需要设置信道有效值

收到