espressif/ESP8266_NONOS_SDK #285: 无法完成编译，提示”make: 对“FORCE”无需做任何事。”

ZhouChenghong, 2019-12-21:
我编译V3.0.1和V3.0版本SDK，在deepin系统上自己搭建环境，使用V5.2.0版本工具链进行编译，提示
“
start...

make: 对“FORCE”无需做任何事。
”
为了排除开发环境的原因，用在文档提供的地址下载的虚拟机，进行编译，提示一样，无法完成编译。
请问是哪里有问题？

ustccw, 2020-01-11:
请确保 `xtensa-lx106-elf-gcc -v` 已在环境变量中，确保路径正确。
问题仍然存在，请提供详细的 log, 和操作过程。

ZhouChenghong, 2020-01-11:
刚刚反思了一下，解决了，报错
“
start...

make: 对“FORCE”无需做任何事。
”
的原因是没有把例程的文件夹复制到SDK根目录，而编译在根目录下的例程，因为编译参数选择了不支持的参数，所以一直编译不成功，现在OK了

closed 2020-01-11 by ZhouChenghong

englianhu, 2022-09-18:
```r
$ sudo make
make[1]: 进入目录“/usr/lib/R/m4”
make[1]: 对“R”无需做任何事。
make[1]: 离开目录“/usr/lib/R/m4”
make[1]: 进入目录“/usr/lib/R/tools”
make[1]: 对“R”无需做任何事。
make[1]: 离开目录“/usr/lib/R/tools”
make[1]: 进入目录“/usr/lib/R/doc”
make[2]: 进入目录“/usr/lib/R/doc/html”
make[2]: 离开目录“/usr/lib/R/doc/html”
make[2]: 进入目录“/usr/lib/R/doc/manual”
make[2]: 对“R”无需做任何事。
make[2]: 离开目录“/usr/lib/R/doc/manual”
make[1]: 离开目录“/usr/lib/R/doc”
make[1]: 进入目录“/usr/lib/R/etc”
make[1]: 离开目录“/usr/lib/R/etc”
make[1]: 进入目录“/usr/lib/R/share”
make[1]: 离开目录“/usr/lib/R/share”
make[1]: 进入目录“/usr/lib/R/src”
make[2]: 进入目录“/usr/lib/R/src/scripts”
make[3]: 进入目录“/usr/lib/R/src/scripts”
make[3]: 离开目录“/usr/lib/R/src/scripts”
make[2]: 离开目录“/usr/lib/R/src/scripts”
make[2]: 进入目录“/usr/lib/R/src/include”
make[3]: 进入目录“/usr/lib/R/src/include/R_ext”
make[3]: 对“R”无需做任何事。
make[3]: 离开目录“/usr/lib/R/src/include/R_ext”
make[2]: 离开目录“/usr/lib/R/src/include”
make[2]: 进入目录“/usr/lib/R/src/extra”
make[3]: 进入目录“/usr/lib/R/src/extra/tre”
make[4]: 进入目录“/usr/lib/R/src/extra/tre”
make[4]: “Makedeps”已是最新。
make[4]: 离开目录“/usr/lib/R/src/extra/tre”
make[4]: 进入目录“/usr/lib/R/src/extra/tre”
make[4]: “libtre.a”已是最新。
make[4]: 离开目录“/usr/lib/R/src/extra/tre”
make[3]: 离开目录“/usr/lib/R/src/extra/tre”
make[2]: 离开目录“/usr/lib/R/src/extra”
make[2]: 进入目录“/usr/lib/R/src/appl”
make[3]: 进入目录“/usr/lib/R/src/appl”
make[3]: “libappl.a”已是最新。
make[3]: 离开目录“/usr/lib/R/src/appl”
#@make install-Linpack
make[2]: 离开目录“/usr/lib/R/src/appl”
make[2]: 进入目录“/usr/lib/R/src/nmath”
make[3]: 进入目录“/usr/lib/R/src/nmath”
make[3]: “libnmath.a”已是最新。
make[3]: 离开目录“/usr/lib/R/src/nmath”
make[2]: 离开目录“/usr/lib/R/src/nmath”
make[2]: 进入目录“/usr/lib/R/src/unix”
make[3]: 进入目录“/usr/lib/R/src/unix”
make[3]: “libunix.a”已是最新。
make[3]: 离开目录“/usr/lib/R/src/unix”
make[3]: 进入目录“/usr/lib/R/src/unix”
gcc -I. -I../../src/include -I../../src/include  -I/usr/local/include -DHAVE_CONFIG_H    -g -O2  -L/usr/local/lib64 -DR_HOME='"/usr/lib/R"' \
  -o Rscript ./Rscript.c
make[3]: 离开目录“/usr/lib/R/src/unix”
make[2]: 离开目录“/usr/lib/R/src/unix”
make[2]: 进入目录“/usr/lib/R/src/main”
make[3]: 进入目录“/usr/lib/R/src/main”
make[4]: 进入目录“/usr/lib/R/src/main”
/usr/lib/R/lib/libR.so is unchanged
make[4]: 离开目录“/usr/lib/R/src/main”
make[3]: 离开目录“/usr/lib/R/src/main”
make[3]: 进入目录“/usr/lib/R/src/main”
make[4]: 进入目录“/usr/lib/R/src/main”
/usr/lib/R/bin/exec/R is unchanged
make[4]: 离开目录“/usr/lib/R/src/main”
make[3]: 离开目录“/usr/lib/R/src/main”
[... 347 lines cut ...]
make[3]: 离开目录“/usr/lib/R/src/main”
make[2]: 离开目录“/usr/lib/R/src/main”
make[2]: 进入目录“/usr/lib/R/src/modules”
make[3]: 进入目录“/usr/lib/R/src/modules/internet”
/usr/lib/R/lib64/R/modules/internet.so is unchanged
make[3]: 离开目录“/usr/lib/R/src/modules/internet”
make[3]: 进入目录“/usr/lib/R/src/modules/lapack”
/usr/lib/R/lib64/R/modules/lapack.so is unchanged
make[3]: 离开目录“/usr/lib/R/src/modules/lapack”
make[3]: 进入目录“/usr/lib/R/src/modules/X11”
/usr/lib/R/lib64/R/modules/R_X11.so is unchanged
/usr/lib/R/lib64/R/modules/R_de.so is unchanged
make[3]: 离开目录“/usr/lib/R/src/modules/X11”
make[2]: 离开目录“/usr/lib/R/src/modules”
make[2]: 进入目录“/usr/lib/R/src/library”
installing packages ...
  building HTML index ...
make[2]: 离开目录“/usr/lib/R/src/library”
make[1]: 离开目录“/usr/lib/R/src”
make[1]: 进入目录“/usr/lib/R/tests”
make[1]: 对“install”无需做任何事。
make[1]: 离开目录“/usr/lib/R/tests”
```
https://github.com/scibrokes/setup-centOS7-DO/issues/2
https://github.com/scibrokes/owner/issues/4

xcguang, 2022-09-19:
NONOS SDK 不再维护，请采用 RTOS SDK