espressif/ESP8266_NONOS_SDK #286: dhcp获取地址后配置静态ip, 等dhcp租期到应用崩溃

hznulilingbo, 2020-01-07:
使用最新的master版本, 应用起来连接ssid并dhcp获取到IP之后, 我先后用wifi_station_dhcpc_stop()和wifi_set_ip_info(STATION_IF, &info)两个接口配置了静态IP. 程序每次都在上一次dhcp租期到期时候崩溃Fatal exception 28(LoadProhibitedCause). 
追踪问题好像在底层socket发包时候崩溃
好像wifi_station_dhcpc_stop()和wifi_set_ip_info(STATION_IF, &info)接口并没有完全释放掉dhcp或者socket栈的缓存

yuanjianmin, 2020-01-13:
该问题已经在最新的master Commit ID(3e85cae4314b7ef80322aa308ae27ba0742c4ab2)上解决，请拉取进行验证。

hznulilingbo, 2020-01-13:
这个问题前几天解决了 和你改得一样
另外dhcp客户端还有一个问题: 市场上有些无线路由器的dhcp服务分配的网关是192.168.*.254这种不是常规192.168.*.1的网关地址 并且renew续租时服务端的ack里面没有下发netmask和gateway的IP, 这样的情况如果按照lwip的代码里面处理renew收到的ack时会先清空之前的配置的netmask和gateway, 并且代码默认把gateway设置成了192.168.*.1, 造成设备网络不通.
目前我把renew时的清空老的netmask和gateway注释掉解决了这个问题, 我觉得renew收到ack时不应该清空老数据
这个问题我查看esp32代码应该也有同样的问题
/**
 * Handle a DHCP ACK packet
 *
 * @param netif the netif under DHCP control
 */
static void ICACHE_FLASH_ATTR
dhcp_handle_ack(struct netif *netif)
{
  struct dhcp *dhcp = netif->dhcp;
#if LWIP_DNS
  u8_t n;
#endif /* LWIP_DNS */

  /* clear options we might not get from the ACK */
  //ip_addr_set_zero(&dhcp->offered_sn_mask);
  //ip_addr_set_zero(&dhcp->offered_gw_addr);
#if LWIP_DHCP_BOOTP_FILE
  ip_addr_set_zero(&dhcp->offered_si_addr);
#endif /* LWIP_DHCP_BOOTP_FILE */

yuanjianmin, 2020-01-14:
按照lwip源码里的逻辑，如果dhcp ack包里不带gateway和netmask，的确会设置一个netmask为255.255.255.0和gateway为192.168.*.1，但是要先确认dhcp ack包里不带gateway和netmask这个行为对不对，我目前还没有遇到过该行为的路由器，你那边有遇到过吗？具体型号是多少？谢谢。

hznulilingbo, 2020-01-14:
MikroTik 951

yuanjianmin, 2020-01-14:
我看了lwip的源码，你可以看一下dhcp discover和dhcp request包里面option55,里面会有客户端想从服务器获取的Parameter List，里面会包含Subnet Mask, Router, Domain Name Server等，如果客户端请求了，服务器没回该字段的话，那就是服务器的问题了。

hznulilingbo, 2020-01-14:
按照dhcp协议是不规范的，但是不能确定市场上所有的无线路由器是否都符合规范，个人觉得还是需要兼容，而且处理这种renew没有掩码和网关的情况把原来的掩码和网关清除掉自己去猜测掩码和网关这样合适吗？

hznulilingbo, 2020-01-14:
我这边改动是注释掉了清除操作。我只是反馈了问题和提出个人建议。

closed 2020-01-14 by hznulilingbo