espressif/ESP8266_NONOS_SDK #292: SSL connection fails with espconn_secure_ca_enable on current master

gilpinheiro, 2020-01-22:
Latest master build always drops connection when espconn_secure_ca_enable is set.

I've reproduced this a few different ways and can confirm that the code worked previously (and correctly validated the server's key was signed by the CA certificate).

When working master@ed302d1eab70f17848e7c7b671bc2ecc1336b851:
```
client handshake start.
espconn_mbedtls.c 662, type[certificate],length[928]
espconn_mbedtls.c 662, type[certificate],length[928]
espconn_mbedtls.c 662, type[private_key],length[887]
espconn_mbedtls.c 662, type[TLS.ca_x509.cer],length[920]
client handshake ok!
```
(I've verified that the server is being properly validated - connection fails as expected if I use a server certificate not signed by the CA certificate)

on master@HEAD build:
```
client handshake start.
espconn_mbedtls.c 662, type[certificate],length[928]
espconn_mbedtls.c 662, type[certificate],length[928]
espconn_mbedtls.c 662, type[private_key],length[887]
espconn_mbedtls.c 662, type[TLS.ca_x509.cer],length[920]
client handshake failed!
Reason:[-0x2700]
INFO connection.c:99 (reconnect_cb) reconnection occurred [UNKNOWN_ERROR?] (-39)
```

May be related to #275 (it looks like maybe he is also using a CA certificate to authenticate)

gilpinheiro, 2020-01-22:
Test code looks like:
```
    bool ca_error = false;
    info("About to start CA_ENABLE");
    ca_error = espconn_secure_ca_enable(1, locate_ca_sector());
    if (ca_error == false)
        error("espconn_secure_ca_enable has failed");

    info("About to start cert_req");
    ca_error = espconn_secure_cert_req_enable(1, locate_ca_sector() + 1);
    if (ca_error == false)
        error("espconn_secure_cert_req_enable has failed");

    info("... starting connection");
    err = espconn_secure_connect(&connection);
```

davydnorris, 2020-01-22:
Yes - I am sure this is related. I have rolled back the mbedtls lib to before the issue I reported and everything works fine. All other updates since then are OK but that change in #275 broke SSL and it's still not fixed

davydnorris, 2020-01-22:
The other thing that they did in that update was re-enable SSL3, which is really dangerous as it has known security flaws