espressif/ESP8266_NONOS_SDK #294: Association fails if network uses ARP caching

flameboss, 2020-01-24:
This problem is described very well by jameswah here:

https://community.meraki.com/t5/Wireless-LAN/ESP8266-and-frequent-reconnects-fail-with-latest-AP-firmware-MR/td-p/12707

> I had a very similar problem on a Mobility Express network. The issue seems to be that the ESP8266 firmware, on receiving a DHCP offer, will send an ARP request to see if anyone is using the same IP. If it gets a reply, it rejects the IP and asks for another. The problem is, at least with ME, the infrastructure responds from its own ARP cache. Then, if the ESP8266 has been on the network recently, it gets an ARP response with its own MAC address! - and barfs. The workaround on ME is to disable ARP caching with `config flexconnect arp-caching disable`, though it's a band-aid. It looks like there is a relevant bug on the Espressif core: https://github.com/espressif/esp-idf/issues/2182 It also seems like other LWIP apps could have the same behaviour.

Here is a commit that fixes the problem and should make associating complete a little faster, too.

https://github.com/flameboss/ESP8266_NONOS_SDK/commit/b84bd6541a3c76adca94f673b401c53704914651

I know Espressif is only fixing critical bugs on this project now, but if someone does not have control of the corporate network configuration to disable ARP caching - this can be a critical problem causing ESP8266 to not work on the network.

flameboss, 2020-01-24:
Just adding some more info... This problem also causes other problems on a network with ARP caching:

1. Devices receive a different IP address every time they associate (minor issue) and it this can cause the network to stop issuing IP addresses when it runs out IP addresses in its configured range.
2. You cannot assign a fixed IP address for the IoT device using DHCP on the network.

Related issue?
https://github.com/espressif/esp-idf/issues/2182

ccrisan, 2020-01-24:
@flameboss 

> Here is a commit that fixes the problem and should make associating complete a little faster, too.

Would that change require rebuilding of parts of the SDK or does it suffice to simply recompile my user code?

flameboss, 2020-01-24:
It requires rebuilding part of the SDK, the lwip library. If you build your app inside the SDK folder, then run these two lines before you do your normal build.

> cd $(SDK_DIR)/third_party/lwip && make COMPILE=gcc
	cp $(SDK_DIR)/third_party/lwip/.output/eagle/debug/lib/liblwip.a $(SDK_DIR)/lib/

ccrisan, 2020-01-26:
@flameboss worked like a charm. Thanks!

davydnorris, 2020-01-27:
I believe this is an issue that's been causing me and several others grief for ages - great find! I have seen this behaviour on quite a few occasions because my devices connect to public Wifi.

@xcguang - can we get this rolled into the SDK please?

eriksl, 2020-04-13:
A DHCP client shouldn't be checking the IP address it receives, that's the task of the DHCP server (which is very much better equipped to do so anyway).

rohansingh, 2022-04-22:
@eriksl The [relevant RFC][1] disagrees. It's not required for a client to check the IP address, but it is recommended:

     The client receives the DHCPACK message with configuration
     parameters.  The client SHOULD perform a final check on the
     parameters (e.g., ARP for allocated network address), and notes the
     duration of the lease specified in the DHCPACK message.  At this
     point, the client is configured.  If the client detects that the
     address is already in use (e.g., through the use of ARP), the
     client MUST send a DHCPDECLINE message to the server and restarts
     the configuration process.  The client SHOULD wait a minimum of ten
     seconds before restarting the configuration process to avoid
     excessive network traffic in case of looping.

I think the problem with ESP-IDF is that, after sending a decline, it doesn't restart the DHCP process and try to get another IP.

[1]: https://www.ietf.org/rfc/rfc2131.txt

flameboss, 2022-04-25:
But then wouldn't that be 2 problems with IDF... 1) the check is not correct because it considers an ARP response for its own mac address as an error, and 2) it doesn't restart DHCP process, as you said.

phildimond, 2023-09-26:
Put

#define CONFIG_LWIP_DHCP_DOES_ARP_CHECK 0

In your main before any #includes.