espressif/ESP8266_NONOS_SDK #302: 如何使用ESP8266_NONOS_SDK作为客户端，连接https的网站。

zouzhe1, 2020-02-17:
**我的根本目的是，要连接一个https的天气api**
然后我在网上只能找到连接http的资料，没有https的资料。
我参照iot_demo里面的SERVER_SSL_ENABLE的函数
在我的http里面，分别调用了以下函数
```
espconn_secure_send((struct espconn *)arg, HTTPS_MESSAGE, os_strlen(HTTPS_MESSAGE));
espconn_secure_set_size(0x01,4096);
espconn_secure_connect(&ST_NetCon);
```
其他都是http的，没动。
我的https请求头如下：
```
#define HTTPS_MESSAGE "GET https://way.jd.com/he/freeweather?city=beijing&appkey=xxxxxxx HTTP/1.1  \r\n" \
"Host:way.jd.com  \r\n"  \
"Connection:close  \r\n\r\n"
```
最终串口输出结果如下循环输出：
```
----ESP_TCP_CONNECT OK!----
esp8266 send OK!
client's data invalid protocol
Error: invalid protocol message
----ESP_TCP_BREAK!----
please start sntp first !
client handshake start.
client handshake ok!
```

zouzhe1, 2020-02-17:
并且全程我也没弄SNTP的代码，不知道为啥莫名其妙说please start sntp first

xcguang, 2020-02-17:
SSL 需要先启动 SNTP，并获取到正确的时间

xcguang, 2020-02-17:
新项目建议采用 RTOS 版本，nonos 差不多放弃更新了