espressif/ESP8266_NONOS_SDK #306: crash in xPortGetFreeHeapSize or lmac.c:508

st0ff3r, 2020-04-16:
in version 3.0.3(8427744) occasionally I get a crash:

reason=2 exccause=28 epc1=0x40100e22 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

dissassembling the code I get the folowing:

I am not sure though how it should be read - I am not sure how this address 40100e22 would be reached.

xtensa-lx106-elf-addr2line 40100e22
/home/xcg/workspace/debug/esp8266_nonos_sdk_core_20180510/app/pp/lmac.c:508

so is it actually in lmac.c code?

40100e10 <xPortGetFreeHeapSize>:
40100e10:       ff3321          l32r    a2, 40100adc <ets_timer_arm_new+0x174>
40100e13:       5228            l32i.n  a2, a2, 20
40100e15:       f00d            ret.n
40100e17:       4c6100          excw
40100e1a:       ff              .byte 0xff
40100e1b:       0658            l32i.n  a5, a6, 0
40100e1d:       06b527          bgeu    a5, a2, 40100e27 <xPortGetFreeHeapSize+0x17>
40100e20:       056d            mov.n   a6, a5
40100e22:       0558            l32i.n  a5, a5, 0
40100e24:       f83527          bltu    a5, a2, 40100e20 <xPortGetFreeHeapSize+0x10>
40100e27:       1678            l32i.n  a7, a6, 4
40100e29:       376a            add.n   a3, a7, a6
40100e2b:       079237          bne     a2, a3, 40100e36 <xPortGetFreeHeapSize+0x26>
40100e2e:       1248            l32i.n  a4, a2, 4
40100e30:       062d            mov.n   a2, a6
40100e32:       447a            add.n   a4, a4, a7
40100e34:       1649            s32i.n  a4, a6, 4
40100e36:       1278            l32i.n  a7, a2, 4
40100e38:       872a            add.n   a8, a7, a2
40100e3a:       0f9587          bne     a5, a8, 40100e4d <xPortGetFreeHeapSize+0x3d>
40100e3d:       ff2781          l32r    a8, 40100adc <ets_timer_arm_new+0x174>
40100e40:       6888            l32i.n  a8, a8, 24
40100e42:       101587          beq     a5, a8, 40100e56 <xPortGetFreeHeapSize+0x46>
40100e45:       1598            l32i.n  a9, a5, 4
40100e47:       0558            l32i.n  a5, a5, 0
40100e49:       997a            add.n   a9, a9, a7
40100e4b:       1299            s32i.n  a9, a2, 4
40100e4d:       0259            s32i.n  a5, a2, 0
40100e4f:       011267          beq     a2, a6, 40100e54 <xPortGetFreeHeapSize+0x44>
40100e52:       0629            s32i.n  a2, a6, 0
40100e54:       f00d            ret.n
40100e56:       085d            mov.n   a5, a8
40100e58:       fffc46          j       40100e4d <xPortGetFreeHeapSize+0x3d>
40100e5b:       89f000          excw
40100e5e:       fe              .byte 0xfe
40100e5f:       3f              .byte 0x3f
40100e60:       000000          ill
40100e63:       fe3180          excw

eriksl, 2020-04-24:
You're sure it's not due to a stray pointer in your own code that corrupts system memory? I've never seen this.

st0ff3r, 2020-04-29:
could it maybe be becouse of functions taking to long time to run... I've noticed wifi_softap_set_config() can take over a second to complete