espressif/ESP8266_NONOS_SDK #308: dev 1163 with no ISR

FatihBAKIR, 2020-04-30:
Hello,

I'm getting the `dev 1163` then wdt reset issue. However, all the resources I can find mention long ISRs.

I am not using any ISR whatsoever (even USART), but I get this error regardless.

Is there any reason this could be happening?

genotix, 2020-07-06:
I have the same issue.

This page says something concerning the power source:
https://www.engineer2you.com/2017/05/esp12e-arduino-programming.html

This is my error
`23:55:13.522 -> SDK:2.2.2-dev(38a443e)/Core:2.7.1=20701000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
23:55:13.522 -> Booting...
23:55:13.522 -> Device Init...
23:55:13.634 -> scandone
23:55:13.634 -> state: 0 -> 2 (b0)
23:55:13.634 -> state: 2 -> 3 (0)
23:55:13.634 -> state: 3 -> 5 (10)
23:55:13.634 -> add 0
23:55:13.634 -> aid 1
23:55:13.634 -> cnt 
23:55:13.634 -> 
23:55:13.634 -> connected with Walmolen 5, channel 1
23:55:13.667 -> dhcp client start...
23:55:14.688 -> ip:192.168.40.57,mask:255.255.255.0,gw:192.168.40.1
23:55:15.049 -> sleep disable
23:55:17.641 -> scandone
23:55:18.684 -> state: 5 -> 2 (b0)
23:55:19.689 -> state: 5 -> 0 (2)
23:55:19.689 -> reconnect
23:55:19.689 -> Connection Failed! Resetting WIFI...
23:55:19.728 -> scandone
23:55:19.728 -> del if0
23:55:19.728 -> usl
23:55:19.728 -> mode : null
23:55:25.801 -> mode : sta(84:0d:8e:72:7a:49)
23:55:25.801 -> add if0
23:55:29.040 -> scandone
23:55:30.083 -> state: 0 -> 2 (b0)
23:55:30.117 -> state: 2 -> 3 (0)
23:55:30.117 -> state: 3 -> 5 (10)
23:55:30.117 -> add 0
23:55:30.117 -> aid 1
23:55:30.117 -> cnt 
23:55:30.117 -> 
23:55:30.117 -> connected with Walmolen 5, channel 1
23:55:30.117 -> dhcp client start...
23:55:30.158 -> ip:192.168.40.57,mask:255.255.255.0,gw:192.168.40.1
23:55:30.158 -> OTA port 8266
23:55:30.158 -> OTA Ready at IP address: 192.168.40.57
23:55:30.158 -> Initializing housenumber
23:55:30.158 -> Setup done.
23:55:40.100 -> pm open,type:0 0
00:06:27.561 -> dev 1163
00:06:35.755 -> 
00:06:35.755 -> ⸮⸮K⸮D	⸮\%⸮⸮u⸮@⸮y+⸮uB	-)z⸮ JTD⸮⸮T\%	)n@⸮	5[
00:06:35.755 -> !Lz JJk⸮C⸮	⸮⸮TX⸮⸮E⸮)⸮T⸮
00:06:35.825 -> SDK:2.2.2-dev(38a443e)/Core:2.7.1=20701000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
00:06:35.825 -> Booting...
00:06:35.825 -> Device Init...
00:06:35.934 -> scandone
00:06:35.934 -> state: 0 -> 2 (b0)
00:06:35.975 -> state: 2 -> 3 (0)
00:06:35.975 -> state: 3 -> 5 (10)
00:06:35.975 -> add 0
00:06:35.975 -> aid 1
00:06:35.975 -> cnt 
00:06:35.975 -> 
00:06:35.975 -> connected with Walmolen 5, channel 1
00:06:35.975 -> dhcp client start...
00:06:36.980 -> ip:192.168.40.57,mask:255.255.255.0,gw:192.168.40.1
00:06:37.401 -> sleep disable
00:06:39.959 -> scandone
00:06:41.031 -> state: 5 -> 2 (b0)
00:06:42.023 -> state: 5 -> 0 (2)
00:06:42.023 -> reconnect
00:06:42.023 -> Connection Failed! Resetting WIFI...
00:06:42.023 -> scandone
00:06:42.023 -> del if0
00:06:42.023 -> usl
00:06:42.023 -> mode : null
00:06:48.114 -> mode : sta(84:0d:8e:72:7a:49)
00:06:48.114 -> add if0
00:06:51.339 -> scandone
00:06:52.430 -> state: 0 -> 2 (b0)
00:06:52.430 -> state: 2 -> 3 (0)
00:06:52.430 -> state: 3 -> 5 (10)
00:06:52.430 -> add 0
00:06:52.430 -> aid 1
00:06:52.430 -> cnt 
00:06:52.430 -> 
00:06:52.430 -> connected with Walmolen 5, channel 1
00:06:52.465 -> dhcp client start...
00:06:52.465 -> ip:192.168.40.57,mask:255.255.255.0,gw:192.168.40.1
00:06:52.465 -> OTA port 8266
00:06:52.465 -> OTA Ready at IP address: 192.168.40.57
00:06:52.465 -> Initializing housenumber
00:06:52.465 -> Setup done.
00:07:02.423 -> pm open,type:0 0
00:36:44.568 -> dev 1163
00:36:52.909 -> B⸮
%dP⸮MT'⸮⸮w⸮⸮)S⸮G⸮ ⸮\5⸮⸮
00:36:52.909 -> ⸮r%	)dС⸮	5Y
00:36:52.909 -> 
00:36:52.909 -> L⸮A⸮	I⸮VX-Cҭ 9)M⸮⸮⸮	⸮⸮`

genotix, 2020-07-08:
I have placed two capacitors over the power of the ESP8285 (ESP-M3) and still have crashes.
The first crash occurred after 27 minutes.
The next time I tested it crashed after 30 minutes.

Not sure what causes this.

genotix, 2021-01-02:
I have again the same issue with no ISR and again an ESP-M3.

Using FastLED library. It seemed to mainly occur when I was calling the FastLED.delay() function which did have a yield(); in there.

Thinking to have resolved the problem by using only FastLed.show() combined with a millis() based timed event showed me not effective enough.

I think I'll be grabbing a ESP8266 and see if the same issue occurs with that too.

genotix, 2021-01-03:
ESP8266 has the same issue.

It appears to be something reported on 20 Mar 2019 too.
https://github.com/espressif/ESP8266_NONOS_SDK/issues/90

I'm thinking there might be a WiFi interrupt executed during the FastLED.show() causing the Watchdog to perhaps falsely "think" the FastLED.show() is taking too long.

I wonder wether the watchdog gets a .feed() on an interrupt.

Note that I have the Wifi OTA scheme setup too.