espressif/ESP8266_NONOS_SDK #309: [SDK 3.0.3] csum err with ESP 8285

ThomArmax, 2020-04-30:
Hello,

I try to flash the the new AT firmware 1.7.3 onto my ESP 8285. I have tried different bins, and addresses, but each time I get
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 816, room 16 
tail 0
chksum 0xef
load 0x00000000, len 0, room 8 
tail 0
chksum 0xef
load 0x00000000, len 0, room 0 
tail 0
chksum 0xef
csum 0xef
csum err
ets_main.c 

```

What is the new flash map to be used for ESP 8285 ?

Thanks

xcguang, 2020-04-30:
How did you flash it? It seems there is no boot.bin in the flash

ThomArmax, 2020-04-30:
@xcguang I have flashed

| file | addr |
|-----------------------|-----------|
| boot_v1.7.bin             | 0x00000 |
| user1.2048.new.5.bin | 0x01000 |
|esp_init_data_default_v08.bin | 0x1fc000|
|blank.bin | 0xfe000 |
| blank.bni | 0x1fe000 |

kriegste, 2020-04-30:
Which flash mode did you use (qio, dout, and so on)? On several modules I only was able to use "dout".

ThomArmax, 2020-04-30:
@kriegste I have no choice, since it's ESP 8285
- SPI SPEED : 40MHz
- SPI MODE (can't change) DOUT
- FLASH SIZE (can't change) 8Mbit
- DoNotChgBin checked

ThomArmax, 2020-05-11:
@xcguang @kriegste Do you have any idea ? Thanks :)

kriegste, 2020-05-11:
Check your flash map. If you really have 8Mbit (= 1MB) of flash, there is no address 0x1fc000 or 0x1fe000.
Also, you need the firmware with "1024" in the name.

closed 2021-08-09 by xcguang