espressif/ESP8266_NONOS_SDK #311: esp-now: How can I find out if a message is encrypted or not?

kriegste, 2020-05-16:
My project switches devices on and off and is not necessarily logged onto a WiFi. So I thought esp-now is worth a try. For security reasons I need encryption so that no unauthorized sender can switch on/off my device. My understanding is that I need to register the possible receivers using

esp_now_add_peer(MAC1, ESP_NOW_ROLE_COMBO, CHANNEL, key, sizeof(key));

Now the problem is, I can still receive unencrypted messages, e.g. broadcast messages. And there is no way I can find out that they actually were unencrypted! So any attacker can easily forge messages that switch on and off my devices. How do I prevent that?

kriegste, 2020-05-16:
Ideally, there should be a flag in the receive callback, e.g.

static void ICACHE_FLASH_ATTR cbNowRecv(uint8_t *mac_addr, uint8_t *pdata, uint8_t len, **bool encrypted**) {
...
}

so we can decide whether we accept the message or not. This way we could still have unencrypted broadcast messages for other uses.

RutgerOlsbergs, 2022-03-31:
Has anything happened with this issue? Just noticed the same behaviour today and it's a major security concern.