espressif/ESP8266_NONOS_SDK #312: wifi_station_scan() callback argument status is FAIL or CANCEL at certain locations

st0ff3r, 2020-05-22:
I have an app connected to an AP and every 16 seconds it scans for wireless networks but recently in one specific location the callback is called with status == FAIL and later CANCEL. After I "reset" the network by calling wifi_station_disconnect() and wifi_station_connect() it works again for a while.

If I move the device to another location it works fine.

When the error start to happen I also get "switch to channel 10" or "switch to channel 1" - and some debugging showed that it was "hanging" at that channel and lost network connectivity.

I can find that string in libnet80211.a but hard to debug without the source.

Any ideas how to debug it?

I can find this references to the same problem, but no solution: https://www.esp8266.com/viewtopic.php?f=6&t=7508 and https://www.esp8266.com/viewtopic.php?f=12&t=4789

The source code is here: https://github.com/nabovarme/MeterLogger

kriegste, 2020-05-22:
Could it be a certain access point in that location which uses a non-standard SSID of some kind?
The only problem related to scanning I remember was an overflow in my code in locations with way too many APs!
Try scanning with your phone.

st0ff3r, 2020-05-22:
There are around 30 access points visible. All have ascii character names.

st0ff3r, 2020-05-24:
tried both active and passive mode scanning - same result

kriegste, 2020-05-25:
You could try scanning one channel only (struct scan_config.channel != 0). If this is ok, try the next channel. Maybe this way you can find out if it really is an AP that is causing this. Or if it is some kind of overflow.

st0ff3r, 2020-05-25:
Yes, when I scan a specific channel that is not channel 10, it works, but it is still loosing network connectivity and prints "switch to channel 10" when I try to scan channel 10. So I guess it is caused by some kind of wireless interference on that channel. But scanning call back should return anyway for the rest of the channels.

I have a BladeRF, maybe I should look onto the specific spectrum?

st0ff3r, 2020-05-26:
tried to check the spectrogram with my sdr. There seams to be something making noise on channel 10: [image] but still, the esp should return to the channel it was connected on before scanning.

st0ff3r, 2020-05-26:
I did a tcpdump on channel 10: it seams that the esp is sending "Tag: DS Parameter set: Current Channel: 10" with no reason. Attached output from wireshark:

[image]

st0ff3r, 2020-06-03:
ESP's appear to honour 802.11 beacons containing CSA even if they are not connected to the AP sending the beacon.

Here is a simple command that sends out beacons containing CSA's
https://github.com/nabovarme/beacon_spammer

This is a severe error in the 802.11 implementation.

st0ff3r, 2021-11-30:
Looking closer to libnet80211.a it looks like its based on FreeBSD net80211 https://github.com/freebsd/freebsd-src/ ... s/net80211

So in a desparate try to force the esp to _not_ sending out CSA while scanning, I patched the function ieee80211_add_csa() in ieee80211_output.o like this:

40214748 <ieee80211_add_csa>:
40214748: f00d ret.n

so it completely skips the adding of CSA.

It worked!
I am not seeing "switch to channel X" anymore if there are no ESP's with unpatched ieee80211_add_csa close.

But a better solution would be if it would not _react_ to CSA's sent from other ESP's acting as an access points, when the client is not connected to that specific access point.

Any chances you could look into it? As it is now, the ESP's connection to an access point can be disconnected by anybody sending "evil" beacon packages.

st0ff3r, 2022-01-30:
thought it was fixed in latest commit (1391d154092f9e0e7702086ab9af1558584cc694) , but unfortunately I still get it:

switch to channel 12
scandone
        -> wifi_scan_done_cb(3fff68d8, 1)

st0ff3r, 2022-01-31:
...but it appears to go away if I patch cnx_csa_fn() to do nothing:

40217830 <cnx_csa_fn>:
40217830:	f00d      	ret.n

actually it works better than patching ieee80211_add_csa() since it doesn't react to CSA's while scanning instead of not sending CSA's (I guess).

What is cnx_csa_fn() supposed to do anyway?