espressif/ESP8266_NONOS_SDK #318: there's no fn to disable hardware interrupts once you start them

sneakyimp, 2020-06-11:
I refer to the [hw_timer](https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/driver_lib/driver/hw_timer.c) functions. You can easily create a timer fn with this code:
```
#include "hw_timer.h";

// interrupt handler fn
ICACHE_RAW_ATTR void my_callback(void) {
  // this function will never stop running once you start the hw interrupt
}

void setup() {
  hw_timer_init(NMI_SOURCE,1);
  hw_timer_set_func(my_callback);
  hw_timer_arm(100); 
}
```
Also, there is no documentation at all abou twhat the second param of hw_timer_init does. And these functions become VERY UNSTABLE if you aren't doing anything in the interrupt handler callback.

flameboss, 2020-06-14:
Call hw_timer_init with second argument = 0; then in your ISR call hw_timer_arm() if you want to keep it going and do not call it if you want to "disarm."

someburner, 2020-07-09:
Here's how I do it:

```c
static void ICACHE_RAM_ATTR hw_timer_disarm_internal()
{
    /* Set no reload mode */
    RTC_REG_WRITE(FRC1_CTRL_ADDRESS, DIVIDED_BY_16 | TM_EDGE_INT);

    TM1_EDGE_INT_DISABLE();
    ETS_FRC1_INTR_DISABLE();
}
```

closed 2021-08-09 by xcguang