espressif/ESP8266_NONOS_SDK #323: system_rtc_mem_read fails intermittently

philrittenhouse, 2020-07-12:
Has anyone seen system_rtc_mem_read() return false? The doc's don't seem to provide any details on what that means other than the command failed.  Can someone from Espressif tell me what conditions will cause it to return false?

I am seeing an intermittent failure of the call and I'm trying to understand what could be causing it to fail. My code always calls it with the same three constant values, so I don't think it is related to the inputs. I'm wondering if it might be a hardware problem or maybe a race condition of some kind. So far I have only seen it on one device so that also makes me think it might be hardware related. 

I'm using the ESP-12F from AI Thinker and our software is built using Arduino.  My code calls system_rtc_mem_read() after waking from a deep sleep to restore state of about 300 bytes. I have tried to reproduce the problem with a simple test case but so far I haven't had any luck.

Any insight would be greatly appreciated.

Best regards,
Phil

davydnorris, 2020-07-12:
Different units or the same one? Could it be bad hardware?

philrittenhouse, 2020-07-13:
I have only seen it a couple times but they were all on the same unit.  It could definitely be bad hardware but even that unit works fine 99% of the time.  I really need more data to make any conclusions but it's very hard to reproduce.