espressif/ESP8266_NONOS_SDK #330: Segfaults with the latest SDK version

slaff, 2020-09-28:
For [Sming](https://github.com/SmingHub/Sming), a ESP8266 framework, we tried [updating](https://github.com/slaff/Sming/issues/62) the NON-OS SDK  to version 3.0.4 (commit https://github.com/espressif/ESP8266_NONOS_SDK/commit/be2f86d30c46267ae4fed66f08138c1881eb71c1). After the update all samples segfault as soon as there is an attempt to connect to an AP. The errors point to ieee80211_setup_ratetable. 
The APs that we used are using 5G frequencies. 

**Can you give us any hints what can cause this problem?**

***** Fatal exception 3 (LOAD_STORE_ERROR)
pc=0x4020884b sp=0x3ffffe30 excvaddr=0x40231434
ps=0x00000030 sar=0x0000001e vpri=0xffffffff
r00: 0x402087f7=1075873783 r01: 0x3ffffe30=1073741360 r02: 0x00000002=         2 
r03: 0x0000000b=        11 r04: 0x3ffee344=1073668932 r05: 0x40231434=1076040756 
r06: 0x3ffee1c4=1073668548 r07: 0x3ffffe30=1073741360 r08: 0x00000001=         1 
r09: 0x3ffea9cf=1073654223 r10: 0x00000016=        22 r11: 0x00000000=         0 
r12: 0x00000002=         2 r13: 0x3ffeea30=1073670704 r14: 0x00000000=         0 
r15: 0x3ffeea30=1073670704 

```
0x400005e1: ?? ??:0
0x4020884b: ieee80211_setup_ratetable at ??:?
0x402087f7: ieee80211_phy_init at ??:?
0x40231434: chip_v6_unset_chanfreq at ??:?
0x401049ea: pp_post at ??:?
0x4010426f: lmacRxDone at ??:?
0x40205c90: ieee80211_setup_rates at ??:?
0x4020c468: cnx_update_bss_more at ??:?
0x4020c485: cnx_update_bss_more at ??:?
0x40209ef7: scan_parse_beacon at ??:?
0x40000072: ?? ??:0
0x4022c727: DefFreqCalTimerCB at ??:?
0x40205d7d: ieee80211_parse_beacon at ??:?
0x4020ab7b: sta_input at ??:?
0x40000f58: ?? ??:0
0x4020a73c: sta_input at ??:?
0x4022c0f7: pp_tx_idle_timeout at ??:?
0x4022b9bb: ppPeocessRxPktHdr at ??:?
```

slaff, 2020-09-28:
The issue might be  related to #320.

davydnorris, 2020-09-28:
Just one thing - ESP8266 is only 2.4G, so it won't see APs on the 5G band - could be faulting because you're connecting to a non existent AP as far as it's concerned

slaff, 2020-09-30:
Ok, the AP is dual-band and it has the same SSID for the 5G and for the 2.4G frequencies. I was able to reproduce the problem also by trying to list all APs without actually trying to connect to any of them. The problem seems to occur when there is a 5G AP. Some of the APs have also band steering. The problem does not occur with previous versions of NON-OS SDK 3.0.x.

davydnorris, 2020-09-30:
OK that's interesting,

I do a wifi scan in my code and am using almost the current commit and have seen no problems. I have also got a dual band AP with band steering, but the SSIDs for the two bands are different.

Does it work if you turn off the 5G band or rename the SSID, and how many APs are you seeing? Could you be running out of memory because the scan list is too large?

kriegste, 2020-09-30:
I just configured my 5GHz AP to use the same SSID as the 2.4GHz one. No problem here. ESP8266 can still scan and connect, no crash.

davydnorris, 2020-09-30:
@slaff  - how many APs do you have in your list? Could you be overflowing a list somewhere?

slaff, 2020-10-09:
> Does it work if you turn off the 5G band or rename the SSID

So far I have noticed the following:
- scanning and listing of APs works without a problem with multiple APs. I had 24 - 30 around me and there was no segfault.
- connecting to 2.4 G AP is segfaulting. At it does not matter if the AP is dual-band or hast band steering.
- if the AP is down segfault is not happening. There must be something that has changed in the WIFI client to AP connection code.

I have to go to the woods this week to find a place without WIFI APs and test there. Living in a city has the disadvantage of having 10-20 APs around you.

kriegste, 2020-10-09:
Which model/manufacturer? Maybe Espressif could test that AP in question themselves.

kriegste, 2021-06-21:
https://github.com/espressif/ESP8266_NONOS_SDK/pull/345