espressif/ESP8266_NONOS_SDK #337: library for SNTP header files.

ameerhazo, 2020-11-10:
Hello everyone, I'm doing a project where I need a timestamp for my payload that I'm sending and for that I thought the SNTP API could help me with this. Unfortunately when building my code, I'm getting the "section `.text' will not fit in region `iram1_0_seg'" error. Is this error even related to me using SNTP in my code?

What library should I include for my Makefile in order for SNTP to be compiled?

noizu, 2020-11-11:
its a code size issue, try a difererent rom layout. like 1mb + 1mb.

*   Keith Brings *
   President, Noizu Labs, Inc.

*    Email: *
    keith.brings@noizu.com

*    Skype: *
    keith.brings
    (718) 618-4265

ameerhazo, 2020-11-16:
@noizu Thanks for the tip :D Got my code to work with your help.

closed 2020-11-16 by ameerhazo