espressif/ESP8266_NONOS_SDK #339: Data received through a serial port can sometimes be lost in a few bytes

fengqing-maker, 2020-12-30:
The serial port for ESP8266 can only receive about 100 bytes at a time. Now I send 1024 bytes to ESP8266 through the serial port, which can only receive about 100 bytes at a time. When I received the last piece of data, there were a few bytes missing, which made me receive incomplete data.

ESP8266的串口每次只能接收100个左右的字节,现在我通过串口给ESP8266发送了1024个字节,串口每次只能接收100个左右的字节,在接收到最后一段数据的时候,有很大会丢失几个字节,导致我接收数据不完整

st0ff3r, 2021-01-05:
You need to set an receive interrupt handler and enable serial receive interrupt. The interrupt handler has to read the buffer and save it.

Do you have some code showing what you are doing?

fengqing-maker, 2021-01-12:
uint8_t recvBuffer[1024] = {0};
uint16_t recvLen = 0;
uint16_t oneFrameSize = 0;

LOCAL void ICACHE_FLASH_ATTR ///////
uart_recvTask(os_event_t *events)
{
    if (events->sig == 0) {
#if  UART_BUFF_EN
        Uart_rx_buff_enq();
#else
        uint8_t fifo_len = (READ_PERI_REG(UART_STATUS(UART0)) >> UART_RXFIFO_CNT_S)&UART_RXFIFO_CNT;
        uint8_t d_tmp = 0;
        uint8_t idx = 0;

        os_printf("\r\n===========[%s %d]uart fifo_len:%d, recvLen:%d\r\n", __FUNCTION__, __LINE__, fifo_len,recvLen );
        for (idx = 0; idx < fifo_len; idx++) {
            d_tmp = READ_PERI_REG(UART_FIFO(UART0)) & 0xFF;
            //uart_tx_one_char(UART1, d_tmp);
            recvBuffer[recvLen+idx] = d_tmp;
        }  
        WRITE_PERI_REG(UART_INT_CLR(UART0), UART_RXFIFO_FULL_INT_CLR | UART_RXFIFO_TOUT_INT_CLR);
        uart_rx_intr_enable(UART0);
        recvLen += fifo_len;
       if(recvBuffer[0] == 0x68 && recvBuffer[3] == 0x68)
       {
           oneFrameSize = recvBuffer[2]<< 8 |recvBuffer[1];
           os_printf("oneFrameSize:%d\r\n", oneFrameSize);
       }
       if(recvLen == oneFrameSize)
       {
           PublishData(recvBuffer, recvLen);
           os_memset(recvBuffer, 0, sizeof(recvBuffer));
           recvLen=0;
           oneFrameSize=0;
       }
#endif
    } else if (events->sig == 1) {
#if UART_BUFF_EN
        //already move uart buffer output to uart empty interrupt
        //tx_start_uart_buffer(UART0);
#else

#endif
    }
}

thanks, st0ff3r.

我用的是NONSDK2.2里面的uart.c，在uart_recvTask里面将串口接收到的数据缓存到recvBuffer里面，串口发送的总长度是oneFrameSize，但是recvLen有时候会小于oneFrameSize，丢失了一些字节。另外我将发送给8266的数据和8266接收到的数据对比了一下，其中0x20变为了0x00。并且0x4b、0xff、0x23、0xff、0xfd这些数据丢失了。 

I use uart.c in NONSDK2.2. In UART_RECVTASK, the data received by the serial port is cached into the recvBuffer. I am using NONSDK2.2 inside of uart. C, inside the uart_recvTask to a serial port receives the data cache to recvBuffer inside, is the total length of the serial port to send oneFrameSize, but recvLen sometimes less than oneFrameSize, lost some bytes. And I will be sent to the data of 8266 and 8266 to receive the data contrast, where 0x20 into  0x00. And  these data 0x4b 0xFF, 0x23, 0xFF,0xfd lost.

leonzhao1994, 2023-10-11:
参考https://www.espressif.com.cn/sites/default/files/documentation/esp8266-technical_reference_cn.pdf
是因为fifo溢出了吗？uart的fifo只有128 Byte