espressif/ESP8266_NONOS_SDK #341: WPS fails if there are multiple APs with the same SSID on same channel

sjanek, 2021-02-18:
1: There are two APs with the same SSID, password and they broadcast on the same channel
2: esp8266 statnion is closer to one of the APs
3: we enable WPS on further AP (weaker RSSI)
4: WPS transaction fails (the device try connect to closer AP):
wps discover [NET]
scandone
WPS: neg start
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt
WPS status: 1 // result in the callback
wifi_wps_disable
state: 5 -> 0 (0)
rm 0 

SDK ver: 3.0.4(9532ceb)

Is it possible to work around this by updating the library or doing something with a function, config etc.?

davydnorris, 2021-02-18:
Sounds like they need to examine the MAC address in the AP list as well when doing WPS. That would be the only way to tell them apart

sjanek, 2021-02-19:
Yes, the libwps.a call the wifi_station_set_config and then wifi_station_connect, I think that the bssid is not set in structure passed to wifi_station_set_config function. Unfortunately, I can't do anything about it myself, the library is closed, the only way is to replace the wifi_station_set_config function while WPS is running, but I also need a BSSID to set it there, which is another problem. I replaced the wps_is_selected_pbc_registrar function with the one from freertos lib, unfortunately, it is not fully compatible and I can not get bssid from there too much. There is no good way to do this without changing the library by espressif.

xcguang, 2021-03-17:
```
....
wifi_set_wps_cb(user_wps_status_cb);
....

LOCAL void ICACHE_FLASH_ATTR
user_wps_status_cb(int status)
{
        struct station_config config;

	switch (status) {
		case WPS_CB_ST_SUCCESS:
			wifi_wps_disable();

                         wifi_station_get_config(&config);
                         config.bssid_set = 1;
                         wifi_station_set_config(&config);

			wifi_station_connect();
			break;
		case WPS_CB_ST_FAILED:
		case WPS_CB_ST_TIMEOUT:
			wifi_wps_start();
			break;
	}
}
```

sjanek, 2021-03-17:
Hi Xcuang, We will check this, but the bssid in example is set after the WPS success, but not when in WPS transaction is processed. build public key startbuild public key finishherewps discover [sjj_G]// so if we have to ssj_G networks then here bssid should be selected to this on which the WPS is enebaledscandoneWPS: neg startscandone Best regards,Szymon Janek Od: xcguangWysłano: środa, 17 marca 2021 12:45Do: espressif/ESP8266_NONOS_SDKDW: sjanek; AuthorTemat: Re: [espressif/ESP8266_NONOS_SDK] WPS fails if there are multiple APs with the same SSID on same channel (#341) ....wifi_set_wps_cb(user_wps_status_cb);.... LOCAL void ICACHE_FLASH_ATTRuser_wps_status_cb(int status){        struct station_config config;         switch (status) {               case WPS_CB_ST_SUCCESS:                       wifi_wps_disable();                          wifi_station_get_config(&config);                         config.bssid_set = 1;                         wifi_station_set_config(&config);                        wifi_station_connect();                       break;               case WPS_CB_ST_FAILED:               case WPS_CB_ST_TIMEOUT:                       wifi_wps_start();                       break;        }}

xcguang, 2021-03-17:
OK, we will also test it and reply here.

ustccw, 2021-03-25:
@sjanek @davydnorris could you please test with the following new library:

[wps-test.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/6204446/wps-test.zip)

sjanek, 2021-03-25:
Hi @ustccw,

looks like it's working now, thank you.

ustccw, 2021-03-25:
Thanks, we will sync the bugfix to GitHub as soon as possible.

ustccw, 2021-04-20:
@sjanek it got fixed on commit: https://github.com/espressif/ESP8266_NONOS_SDK/commit/b923ccce464b3d0105789f386727551ae6e059f8
please `git pull` the latest master and test.  

CW.

closed 2021-08-09 by xcguang