espressif/ESP8266_NONOS_SDK #348: DHCP gateway problem after first refresh (leasetime) (STA mode)

atnel, 2021-07-12:
Hi,

I'm using NONEOS AT SDK v.3.0.4. There is small but BUG, when DHCP refresh data from router in STA mode after lease time. 

FISRT time after we connect to router - DHCP get right data - especially Gateway. In my net there it's: 192.168.2.2

SECOND - allways after DHCP refresh data from router, my Gateway become value: 192.168.2.1 (wrong). 

IP and MASK are always ok.

Please try set 2-3 minutes lease time in your router and check this. I have tested this on many routers and many esp8266 modules. 

Mirek

przemyslawzygmunt, 2021-08-05:
It looks like I have the same problem.

przemyslawzygmunt, 2021-08-05:
After connecting to WiFi: ip: 192.168.100.209, mask: 255.255.255.0, gw: 192.168.100.2

Lease renewal:
ip: 192.168.100.209, mask: 255.255.255.0, gw: 192.168.100.1

The problem is particularly troublesome for those users who have a short lease time set and a gateway address that does not end with 1.

atnel, 2021-08-06:
> The problem is particularly troublesome for those users who have a short lease

If I did not have a short lease time set, I would not notice (probably like many people) that this problem exists - or I would look for a problem longer - why my device stops working in certain situations, e.g. after 24 hours ;)

lesny8, 2021-08-06:
There is one more strange case that changes the mask too.

Once connected, an address is assigned
ip: 10.0.4.13, mask: 255.255.255.0, gw: 10.0.4.1

and when renewing the lease, the mask and gate change to
ip: 10.0.4.13, mask: 255.0.0.0, gw: 10.0.0.1

atnel, 2021-08-07:
So for now, there is only one workaround is for that: using static IP only

przemyslawzygmunt, 2021-08-12:
We have verified that this problem occurs from version >= 3.0.1

sq9vgx, 2021-08-12:
I have the same problem.
After connecting to my WiFi network:
ip:10.4.55.93,mask:255.255.255.0,gw:10.4.55.254

when trying to renew address, the mask and the gate take wrong values:
ip:10.4.55.93,mask:255.0.0.0,gw:10.0.0.1

Goral64, 2021-08-12:
I have the same problem :(
After connecting to my WiFi network:
`ip:10.0.144.26,mask:255.255.255.0,gw:10.0.144.2`
when trying to renew address, the mask and the gate take wrong values:
`ip:10.0.144.26,mask:255.0.0.0,gw:10.0.0.1`

kriegste, 2021-08-12:
Could be this
https://github.com/lwip-tcpip/lwip/commit/a8e8ce41080862fd03094a6591798ca0948ef695

przemyslawzygmunt, 2021-08-12:
After a little investigation, I found that the DHCP client was incorrectly parsing the ACK frame when renewing the lease. In version 3.0.0, this problem did not occur because when renewing the lease, the dhcp_handle_ack (netif) function; was not invoked, which was changed in v3.0.1

https://github.com/espressif/ESP8266_NONOS_SDK/commit/959a873f21f06b28fe1156f765b6ccce672465b4#diff-c6c5e739ad3c7a974403ee2df2e4710a220acd785d116f875091cc3f8751091cc3

https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/third_party/lwip/core/dhcp.c#L1746

Now I analyze the parser why it can't handle ACK on renewal ....

xcguang, 2021-08-13:
Could you try AT+RESTORE？

przemyslawzygmunt, 2021-08-13:
The commit pointed to by @kriegste does not fix this problem. This sets the gate to 0.0.0.0 when renewing. I have submitted a PR with a working patch.

kriegste, 2021-08-13:
Unfortunately (fortunately...) I cannot reproduce the error. But the netmask, too, seems to be set to a default. That should be fixed as well.

przemyslawzygmunt, 2021-08-13:
He will also prepare a fix for the mask.

przemyslawzygmunt, 2021-08-13:
I added an amendment regarding the mask.

kriegste, 2021-08-16:
Thanks. Let's see what the testers say...

xcguang, 2021-08-17:
Hello,
Could you provide the router manufacturer and model number, which we can reproduce the issue?
If there are multiple routers, please provide them all, and we can check if we have one.

atnel, 2021-08-17:
I have tested it on MikroTik routers. I think it's not a problem with any kind of routers, but only if you set in your router very short lease time for testing purposes, for example 2-3 minutes.

[image]
here is leasetime config in my router for 3 min

xcguang, 2021-08-19:
@przemyslawzygmunt Thanks very much for your PR https://github.com/espressif/ESP8266_NONOS_SDK/pull/349.

@atnel @kriegste @Goral64 @lesny8 May I ask if this PR https://github.com/espressif/ESP8266_NONOS_SDK/pull/349 is helpful for you? Sorry for that we cannot reproduce this issue on more than five router.

przemyslawzygmunt, 2021-08-19:
Yes, this PR resolve the problem. Confirmed by several of our users.

Goral64, 2021-08-19:
I confirm, problem solved :) Thanks

sq9vgx, 2021-08-19:
I also confirm. The problem has been solved.

lesny8, 2021-08-19:
Yes, this hotfix fixed the problem. Thanks @przemyslawzygmunt

atnel, 2021-08-19:
hotfix  OK

eriksl, 2021-08-28:
The patch suggests the issue only shows when the DHCP server does not supply a netmask (which is _bad_ imho). So that's probably why I don't see it.

przemyslawzygmunt, 2021-08-28:
Mask or IP address

closed 2021-08-31 by wujiangang

eriksl, 2021-09-30:
Even more sounds like a completely broken DHCP server :-(