espressif/ESP8266_NONOS_SDK #353: SDK functions which require interrupts to be deactivated

kriegste, 2021-11-03:
I have a project which uses softap mode and has a lot of uart traffic. I noticed quite some instability there (hardware wdt resets @ 0x40000F68).
I noticed, that in other projects people were disabling interrupts around calls of e.g. wifi_station_connect. I tried that, too, and it helped (a bit). Now I wonder which other functions require interrupts to be deactivated? Is there an official list?

What about...
wifi_station_disconnect
wifi_station_set_config
wifi_softap_set_config
wifi_set_opmode
spi_flash_erase_sector
spi_flash_write
?

eriksl, 2022-01-03:
I never deactivate interrupts before entering SDK functions. Often a fast pacing interrupt is running for PWM and no problem there. Also there are interrupts running for I2S, GPIO and UART.

Are you implementing the "best practice" concept where an interrupt handler is handled in as little instructions as possible, only setting a flag or pushing a task call and then the actual code is run from the main thread?

kriegste, 2022-01-03:
I found out that in the particular project I must not use the UART rxfifo full interrupt and SoftAP at the same time. The serial data comes in quite fast and for some reason those crashes happened from time to time.
Now I do no longer use that particular UART interrupt and the code is much more stable.

eriksl, 2022-01-04:
I am not using SoftAP so I can't comment there. Is this only with the latest SDK version or was that before as well?

I've been fighting for years with the "pin" (GPIO) interrupts. If you manage to get a few in short succession (e.g. if the button isn't debounced in hardware), you'll get a crash. I now have seem to have found a mode that can cope though. Nothing of this is mentioned in the documentation, of course and it definitely needs to be done way different than the examples. My UART ISR looks like this:

`
iram static void uart_callback(void *p)
{
    unsigned int uart0_int_status, uart1_int_status;

    ets_isr_mask(1 << ETS_UART_INUM);

    uart0_int_status = read_peri_reg(UART_INT_ST(0));
    uart1_int_status = read_peri_reg(UART_INT_ST(1));

    if(uart0_int_status & (UART_RXFIFO_TOUT_INT_ST | UART_RXFIFO_FULL_INT_ST)) // data in input fifo of uart0
    {
        stat_uart0_rx_interrupts++;
        enable_receive_int(0, false); // disable input info data available interrupts while the fifo is not empty
        dispatch_post_task(0, task_uart_fetch_fifo, 0);
    }

    if(uart0_int_status & UART_TXFIFO_EMPTY_INT_ST) // space available in the output fifo of uart0
    {
        stat_uart0_tx_interrupts++;
        enable_transmit_int(0, false); // disable output fifo space available interrupts while the fifo hasn't been filled
        dispatch_post_task(0, task_uart_fill_fifo, 0);
    }

    if(uart1_int_status & UART_TXFIFO_EMPTY_INT_ST) // space available in the output fifo of uart1
    {
        stat_uart1_tx_interrupts++;
        enable_transmit_int(1, false); // disable output fifo space available interrupts while the fifo hasn't been filled
        dispatch_post_task(0, task_uart_fill_fifo, 1);
    }

    // acknowledge all uart interrupts

    clear_interrupts(0);
    clear_interrupts(1);

    ets_isr_unmask(1 << ETS_UART_INUM);
}
`

kriegste, 2022-01-04:
Even just disabling the interrupts in the isr and clearing the fifo in an os task before re-enabling them resulted in crashes (with SoftAP mode active).

For the GPIOs I use a simple timer. Checking every 50 ms is sufficient. In one project I have to poll the ADC pin (used for another button) as well, so I do this all in one go in the timer.