espressif/ESP8266_NONOS_SDK #354: Cant connect to ap with latest master

st0ff3r, 2021-11-15:
after upgrading sdk from 20defb6e074b74a433f1f362303a94cdb25480dd to 1fb88a57dcb3254eb5ed429e2c4d40b872a18937 and also to latest wifi_set_event_handler_cb callback returns REASON_MIC_FAILURE.

Also I cant connect wireless clients to the esp8266 access point. Get password incorrect.

Have to make some more detailed logs...

davydnorris, 2021-11-17:
This got broken in the last commit - @kriegste picked it up a few days ago. I've mentioned your issue in the commit comments

st0ff3r, 2021-11-18:
ok cool... lets hope and wait (until they'll free the source code)... ;)

closed 2022-01-19 by wujiangang

ustccw, 2022-01-19:
@all it got fixed. please check.

kriegste, 2022-01-19:
Yes, thanks. The new version seems to be much bigger. We just lost 1,3 kB of heap. :(

ustccw, 2022-01-19:
@kriegste I'm not sure how you compare heap memory, and which two commits you compared.
In my test, latest master commit: https://github.com/espressif/ESP8266_NONOS_SDK/commit/260a60637c1c828a501be549cb7fa95190b26bbc
 `AT+SYSRAM?` returns `54904`, after got ip, it returns `54824`.

And before compiler switchs, i tested commit: https://github.com/espressif/ESP8266_NONOS_SDK/commit/20defb6e074b74a433f1f362303a94cdb25480dd
 `AT+SYSRAM?` returns `55216`, after got ip, it returns `55104`.

There is indeed a loss of about 300 bytes of heap memory.  
This maybe due to internal compiler switched, from xcc to gcc 5.2.0, due to xcc compiler license expired.

kriegste, 2022-01-19:
I am comparing the last working commit before the compiler change to the current commit. Using my own application (not the AT firmware). But nothing has changed except the SDK.
I will try to analyze the problem more accurate, which is a bit harder since now all the libs were modified. Btw, I compile lwip and mbedtls by myself.

davydnorris, 2022-01-20:
> 
> There is indeed a loss of about 300 bytes of heap memory. This maybe due to internal compiler switched, from xcc to gcc 5.2.0, due to xcc compiler license expired.

That's quite an old gcc version - any plans to move to more recent?

Also is there any chance Espressif would consider letting external devs join the maintenance for NonOS, like they have for the RTOS SDK? so many of us still use and need the NonOS SDK

ustccw, 2022-01-20:
@davydnorris 
The `ESP8266_NONOS_SDK` is an earlier version of the software, which has stopped adding new features on December 2019. It is not recommended for customers. See [support policy](https://github.com/espressif/ESP8266_NONOS_SDK#support-policy-for-esp8266-nonos) for more details.
**It is recommended that the customer evaluate and select the new version for [esp-at](https://github.com/espressif/esp-at).**

[ESP8266 IDF AT](https://github.com/espressif/esp-at/tree/release/v2.2.0.0_esp8266), that is the part of common project for esp-at, which is the esp-at software version mainly maintained by espressif, has the following advantages:

1. Richer features and more complete documents. `ESP8266 IDF AT` supports IPv6, MQTT, HTTP and other features, and provides pretty rich [documents](https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp8266/index.html).

2. High priority technical support, continuous repair and update, high activity.

3. Compatible with multiple espressif's chips. [esp-at](https://github.com/espressif/esp-at) supports ESP8266 series, ESP32 series, ESP32-C series, ESP32-S series and new products to be released in the future. It can:

  - (1) A set of applications based on esp-at can be compatible with multiple chips of espressif, avoid maintaining multiple sets of applications for different chips at the same time, and reduce the maintenance cost.

  - (2) In the future, it can easily switch to other espressif's chips, so as to improve flexibility, reduce dependence on a single chip and shorten the time to market.

Above, we hope you can evaluate and consider the selection of `ESP8266 IDF AT`.

If you still need to use `ESP8266 NonOS AT` at for historical reasons, we will also provide you with the required technical support.

Thank you for your attention to ESP!

davydnorris, 2022-01-20:
@ustccw why would we want to use the AT project when we are using the ESP8266 as a stand alone MCU??

The massively popular Arduino port uses this NonOS SDK - how do you propose they use the AT project instead?

I do not think you really understand how the NonOS SDK is actually used - very few people use it for the AT firmware. It's almost exclusively used for developing stand along IoT applications and it's very popular