espressif/ESP8266_NONOS_SDK #365: [SDK 3.0.5][WROOM-02] ESP freeze AT+CIPSEND at high baudrate

ThomArmax, 2022-06-21:
Hello,

I am trying to migrate from SDK 2.0.0 to 3.0.5 because the ESP crash in some WiFi environment.
I'm using the AT UART at high speed (1500000) and I'm getting into trouble with the latest AT firmware (SDK 3.0.5, AT 1.7.5), where the ESP completely freezes after few minutes when sending UDP data (which I don't have with SDK 2.0.0, AT 1.3.0)

After a while, the ESP just does not reply `Recv x bytes\r\nSEND OK`  to `AT+CIPSEND` command, eg : 
```
AT+CIPSEND=0,72,"SOME IP",<PORT>

OK
>
<MY DATA HERE>
```

No more logs into the debug UART, any other AT command don't get any reply from the ESP. 
I have scoped the UART RX/TX to confirm that the ESP does not reply.

If I lower the UART speed to 115200, it looks like to work.

xcguang, 2022-06-22:
Recommend https://github.com/espressif/esp-at/tree/release/v2.2.0.0_esp8266

About not "Recv x bytes\r\nSEND OK",  I guess the data is missing from MCU, you can enable hardware flow control when using high rate and try it again.

ThomArmax, 2022-06-22:
@xcguang thanks for your answer. I already have hardware flow control enabled. The CTS pin is always high on ESP side. My data are well sent after receiving `OK\r\n>`. In parallel I'm testing the ESP-AT version, which is quite different.

xcguang, 2022-06-22:
the difference between nonos-AT and ESP-AT, please refer to https://docs.espressif.com/projects/esp-at/en/release-v2.2.0.0_esp8266/AT_Command_Set/AT_Command_Set_Comparison.html