espressif/ESP8266_NONOS_SDK #374: mdns stop respond after 2-3 minutes

rootmax84, 2023-04-12:
So.. I use this code for test, no matter what sdk version, i try 2.2.1, 3.0.5, 3.0.6 same behaviour

>static struct mdns_info *mdns;
  static void ICACHE_FLASH_ATTR startMDNS(struct ip_addr ip) {
  mdns = (struct mdns_info *)os_zalloc(sizeof(struct mdns_info));
  mdns->host_name = "espTest";
  mdns->server_name = "http";
  mdns->server_port = 80;
  mdns->ipAddr = ip.addr;
  mdns->txt_data[0] = "model=test";
  espconn_mdns_init(mdns);
}

esp respond for their name for a while, than stops responding. If i run mdns-scan from debian esp start responding again for 2-3 minutes and than stops again. It's a bug or feature? :)

fvpalha, 2024-05-08:
Hi @rootmax84 ,
Do you resolved this issue?

rootmax84, 2024-05-08:
Hi, @fvpalha, nope and i think it's unlikely to be fixed.