espressif/ESP8266_NONOS_SDK #381: AT+CWJAP? wrong RSSI returned in AT 1.7.6

JAndrassy, 2024-07-20:
it returns two's complement value of the RSSI. 194 instead of -62. (256 - 62 = 194)

uzi18, 2024-07-21:
Try https://github.com/loboris/ESP8266_AT_LoBo

JAndrassy, 2024-07-21:
@uzi18 what is the point of your comment? Nobody asked you anything. This is an issue tracker, not a forum. I reported the error so it can be fixed. I don't seek a workaround. I could use for example 1.7.4. It would much better than the old Lobo build. 
I wrote and maintain an Arduino library which communicates with the AT firmware and I discovered this error while testing some changes in the library with the latest AT firmware. So I report it so users of my library don't have to deal with it if possible.

milan-stava, 2025-02-26:
Hi, yes, JAndrassy is true, Wrong RSSI in AT+CWJAP_DEF? response (while AT+CWLAP returns the correct value). Could you fix it in the new release, please?